timurrrr/RaceChronoDiyBleDevice #2: waiting for data

zn6aura, 2021-03-16:
Trying my hand at this. but I can't seem to get it to work. checked my wiring 10 times over used a multimeter. using the connector behind the headunit with a toyota harness. 

racechrono is able to connect with bluetooth to the device but never receives any data. have confirmed can is working with an obdlink. 

what would be the best way to troubleshoot this?

thank you for all the information

zn6aura, 2021-03-17:
Okay so I figured it out, I ordered a 3 pack of the mcp2515 following the link in this github. 

Two of the chips are labeled 17200TW. Both of those do not work. I tested the third chip when realizing it was labeled 20239RA. And it worked on its first try... 

So for anyone else trying this out avoid the chips with 17200TW. They look identical but only one works. 

I'm sure there are probably other variations of this as I now see small differences on amazon from different sellers.

closed 2021-03-17 by zn6aura

timurrrr, 2021-03-17:
Yuck! Sorry this happened. For me, 3/3 worked fine.

As long as at least one works fine, I think $9 is still a good price :D

zn6aura, 2021-03-17:
Yeah, no worries, thanks for the hard work on this project. its working great now