timurrrr/RaceChronoDiyBleDevice #5: Add ability to query OBD PIDs and wrap them into fake CAN PIDs

timurrrr, 2021-05-16:
See https://racechrono.com/forum/discussion/2039/obd2-splitter for a relevant discussion.

At least on Subaru BRZ / Toyota 86 / Scion FR-S, I was unable to find some PIDs (e.g. ambient temperature, fuel level) in the CAN bus broadcasts. As those are "slow" channels anyways, it's fine to just query them via OBD protocol, and send them to RaceChrono, pretending these are CAN messages.

alancyau, 2024-06-09:
In RaceChrono, can you add your OBD reader under the "OBD-II readers" section in addition to the "Other devices" section? That way you can still query standard sensor data such as intake air temp.

[image]

timurrrr, 2024-06-16:
@alancyau The device would need to implement both the ELM427 (?) protocol and the RaceChrono DIY protocol. That might be a non-trivial amount of work, and I'm not 100% sure that RaceChrono will correctly handle connecting to the same device over two protocols at the same time.

Another option is to make the CAN reader translate "fake" CAN IDs to periodic OBD requests. I've successfully done that in the past, but didn't get around to publishing the code. I need to clean it up a bit first.