timurrrr/RaceChronoDiyBleDevice #6: Add support for ESP32

jawillis, 2022-06-22:
ESP32 boards are cheap and plentiful.  Hardware projects such as https://github.com/MagnusThome/RejsaCAN-ESP32 would be a ready solution without need for a power supply and external CAN transceiver.
I'm willing to put development effort into this, if you can provide some direction towards a method that would be acceptable for a PR.

timurrrr, 2022-06-29:
Hello jawillis@, and thanks for your recent donation!

In fact Magnus sent me a couple of his boards for testing, so this was something I was already considering.
Unfortunately, I also caught COVID recently, which shifted all my plans.
I will certainly appreciate it if you can help! :)

Take a look at the current source code for this project
https://github.com/timurrrr/RaceChronoDiyBleDevice/blob/master/RaceChronoDiyBleDevice.ino
and how it depends on <RaceChrono.h> provided in this library:
https://github.com/timurrrr/arduino-RaceChrono/blob/master/src/RaceChrono.h

It would be ideal to try to update the `arduino-RaceChrono` library in such a way that the "interface" between the library and the project (`RaceChronoDiyBleDevice.ino`) remains unchanged.
Depending on whether you're compiling it for nRF528** or ESP32 it could automatically pick the right implementation using something like `#if defined(ARDUINO_ARCH_ESP32)`. I presume there should be a similar macro defined by some of the nRF headers, try `NRF52832_XXAA` and `NRF52840_XXAA`?
It might be hard/impossible to preserve the exact "interface" though, so if the underlying BLE libraries for nRF and ESP are very incompatible, we can try to update the "interface" in a way that works with both of them.

So maybe for your initial PR, just rewrite whatever is needed in the .cpp files, while trying to keep the changes in `RaceChrono.h` to the minimum. Then I can take a look at the changes and see how best to organize the code to build the right variant automatically.

Thanks for your help!

jawillis, 2022-06-29:
You're very welcome.  Thanks for putting a great project out there for the community!  I've already built 2 of these for myself and a friend for our BMW e46 track cars.  It was great to be able to just assemble, upload the code and have it work.  I hope you have a speedy recovery from COVID!

It looks like `#if defined(ARDUINO_ARCH_NRF52)` is the macro we're looking for.  

I've been looking more into the `arduino-RaceChrono` code, and I agree that almost all the work will be concentrated there.  I've been thinking of a something basically like [this](https://github.com/MagnusThome/Racechrono-LIVE-telemetry-data-export/blob/main/main/rcmonitor.ino)

There are some other features I would like to include eventually to this project overall, such as multiple canbus interfaces and analog inputs, but I shouldn't get ahead of myself here...

timurrrr, 2022-07-01:
Oh nice, the `rcmonitor.ino` file is a good example of how to write code to "automatically" pick either `<bluefruit.h>` or `<NimBLEDevice.h>` depending on availability.

Cool story about the E46, I'm very happy to hear it was useful for you and your friend :)
Feel free to also send a PR to add a wiki page with common CAN IDs and RaceChrono equations.
Did you figure out how to log the brake pressure data? I think I heard E46's don't have those sensors.
Or is _that_ why you want to add analog inputs?

jawillis, 2022-07-01:
The e46 does have brake pressure sensors, but those messages are only available on a different bus between the abs and engine computers.  That actually why I'd like to add another can interface.  Some people plumb in extra brake pressure sensors to log that data, but it seems more practical to just add another mcp2515 to the project.  The analog inputs would be for engine oil pressure, trans/diff temps, etc.  
I am working on the wiki page for the CAN ID's and equations.  I have already created an e46.h file.

jawillis, 2022-09-03:
I think this can be closed now that ESP32 support has been added.

closed 2022-09-03 by jawillis