timurrrr/RaceChronoDiyBleDevice #10: Test failed on 2022 brz zd8

luzaowei88, 2022-09-17:
Test failed on my 2022 brz
I followed the tutorial exactly, but still failed to get data on my 2022 brz.
"Waiting to fetch data" shows Racechrono both on ios and Android
[https://photos.app.goo.gl/C3WRRwmvh5xWLbzg9](url)

Does this project not support the 2022 brz zd8 ?
thanks.

timurrrr, 2022-09-19:
Do you get any data at all? Try this program instead:
https://github.com/timurrrr/arduino-CAN/tree/master/examples/CanBusMonitor
and watch Serial Monitor in Arduino IDE.

If you got multiple MCP breakout boards, try another one.
Did you do the quartz mod? If not, you need to update the code to match the frequency of the quartz you actually have.

Advice based on my experience: don't use those screw terminals, you're millimeters away from causing a short circuit in the CAN network of your car. Ask me how I know 🥲 To avoid repeating the mistake, I soldered a JST connector to the board and the CAN adapter.

timurrrr, 2022-09-19:
I haven't personally tried on a 2022 BRZ but I'm 99% sure that it works. 2022 BRZ and 2022 GR86 are basically the same car.

luzaowei88, 2022-09-19:
hi, timurrrr, thanks for ur replay.
I've:
1. Changed origin 8hz quartz to 16hz on MCP2515.
2. Fixed error MCP2515.cpp: In member function 'boolean MCP2515Class::setFilterRegisters(uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, bool)' ,etc.
3.  Conected J1 on the MCP2515 to provide a 120 ohm resistor.

Now i am preparing my macbook to try ur canbus monitor and ordered an JST connector , will let you know later.