timurrrr/RaceChronoDiyBleDevice #11: Throttle position went up while braking for Gen 2 BRZ

yzhan1, 2022-11-14:
Hi Timur,

Thanks for putting the database together. I was able to use it for my THW track day and it was super helpful!

One issue I found is sometimes when I brake, the throttle position captured would go up too even though I didn't touch the throttle at all. You can see the video [here](https://youtu.be/A40YHt9TG8o?t=29) when I entered T3. I didn't downshift so there shouldn't be throttle movements at all.

I'm new to CAN bus data logging so I'm not sure what could've caused this. Is this because of a bad equation/PID/RaceChrono/CAN adapter? I used a CAN adapter I purchased online which is specifically designed for Gen 2 BRZ and GR86.

Any help is appreciated!

timurrrr, 2022-11-14:
Hi Yaoming!
The underlying CAN bus data protocol makes it pretty much impossible for any wiring issue to cause wrong data to be received — it has checksums and stuff. It would be a lot more likely to just _stop receiving data_ than to get wrong data.

In your video, I see the throttle pedal data going smoothly from 0% to 15% slightly after you start braking, and smoothly going back to 0% pretty much in sync with how you release the brakes.

My hunch is that you did in fact touch the throttle pedal without realizing it. 
The throttle pedal in these cars is pretty light, so it's hard to notice when it's only lightly pressed, and 15% requires very little force.
I guess at this point we don't have good data to prove that this is what actually happened (or to prove that it definitely didn't happen)...

If you see this happening often, you might consider using a foot cam to see what's going on down there.

timurrrr, 2022-11-14:
Also minor unrelated tip: since you already have the steering angle gauge and you see the steering wheel in your video, you should be able to sync the data to the video more precise.
In your video, the data is a bit behind the video.

yzhan1, 2022-11-14:
Hmmm, interesting... I checked the videos from some other laps and they don't seem to have this issue sometimes, so actually could be a user error :) Thanks for pointing this out!

Re: video sync - I tried using RaceChrono to sync the GoPro videos for me but unfortunately, my phone just doesn't have enough storage space needed so I had to manually sync in iMovie which is much more error-prone :(

closed 2022-11-14 by yzhan1