timurrrr/RaceChronoDiyBleDevice #13: Works with esp32s3

RayDutchman, 2023-10-03:
works with esp32s3. I excluded the CAN.h library but used the TWAI library of the esp32 instead, because it already has a built in CAN controller, no reliability issue at all. And the transceiver I use is SN65HVD230 as recommended by Expressif. This combination is very economic, I just got all the hardware for less than 50 rmb which is about 7 USD. I also added codes to measure one analog reading(the oil pressure) and feed it to the CAN-bus and racechrono with a fake CAN id. Much apperiaction to your work!

RayDutchman, 2023-10-03:
I'm curious about why you wanted to devide the can-data received, because I changed all the deviding number to 1 and there's no apparent issue. Also I wants to figure out what's delay of the transmission process, the loop in my codes takes less than 1ms but apparently the delay is more than that.

timurrrr, 2023-10-05:
I don't remember, I was dealing with some performance/stability issues early in the development of this project. It eventually turned out to be a hardware issue (unstable voltage on the power to the MCP chip). It's very much possible that it's no longer needed.

vilkaslabaipilkas, 2025-01-13:
Hi @RayDutchman, how could I contact you?